P0cL4bs/WiFi-Pumpkin-deprecated #471: DHCP settings unchangeable after first starting

minhducsun2002, 2019-02-17:
It is no longer possible to change DHCP settings once the AP has been started **at least once** . The only way is to restart.

[image]

System details :
- Debian Stretch
- Kernel version `4.20.0-8.2-liquorix-amd64`
- Running bare-metal (no virtualization)
- Network card:
  * `wlan0` (Qualcomm Atheros AR9285 bundled with internal AR5B195) for Internet connection
  * `wlan1` (TP-Link TL-WN722Nv2 with RTL8188EUS, driver [here](https://github.com/kimocoder/rtl8188eus)) hosts the AP.

jepunband, 2019-03-08:
same. no internet on client once connected.

closed 2019-03-08 by mh4x0f

jepunband, 2019-03-14:
didn't solve still the same.