P0cL4bs/WiFi-Pumpkin-deprecated #472: No internet on the client

minhducsun2002, 2019-02-17:
As the title says.

Proof : 
[image]

Reproducing : 
- Get a clean Wifi-Pumpkin installation and fire it up
- Choose the desired interface, **change nothing else** and then start the AP.
- Connect to the newly created AP by whatever device.

System configuration : identical to #471.

jepunband, 2019-03-08:
Hi i'm having the same issue. Started the AP, with proxy server enabled. Client can get ip but there is no internet connection.

mh4x0f, 2019-03-08:
@minhducsun2002 which proxy is activated ?  start the AP again and send me the output command: 
``` sh 
sudo iptables -t nat -L
```

mh4x0f, 2019-03-08:
@minhducsun2002 install this:
``` sh
sudo pip install pyOpenSSL
```

minhducsun2002, 2019-03-08:
```
cipher@magister-mentium [0] ~ $ sudo pip install pyOpenSSL
[sudo] password for cipher: 
Requirement already satisfied: pyOpenSSL in /usr/lib/python2.7/dist-packages (19.0.0)
```

mh4x0f, 2019-03-08:
start the AP send me the output of the command: 

> sudo iptables -t nat -L

minhducsun2002, 2019-03-08:
```
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination         
REDIRECT   tcp  --  anywhere             anywhere             tcp dpt:http redir ports 10000
REDIRECT   udp  --  anywhere             anywhere             udp dpt:domain redir ports 53

Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination         
MASQUERADE  all  --  anywhere             anywhere            

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination 
```

@mh4x0f this?

minhducsun2002, 2019-03-08:
```
cipher@magister-mentium [0] ~/Projects/hestia $ sudo wifi-pumpkin 
:0: UserWarning: You do not have a working installation of the service_identity module: 'cannot import name opentype'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
Loading GUI...
WiFi-Pumpkin Running!
libpng warning: iCCP: known incorrect sRGB profile
[*] Configuring Static AP Mode...
[*] Enable forwarding in iptables...
-------------------------------
AP::[PumpkinAP] Running...
AP::BSSID::[50:3e:aa:ae:a0:c8] CH 11

[*] Loading debugging mode
[*] Current Session::ID [NjMwMTU=]
[*] Sharing Internet Connections with NAT...
[New Thread 5609 (StaticHostapd)]
[New Thread 5613 (DNSMasq Server)]

[*] TCPProxy running on port 80/8080:

TCPProxy::ftp               status:On
TCPProxy::hexdump           status:On
TCPProxy::kerberos          status:On
TCPProxy::httpCap           status:On
TCPProxy::imageCap          status:On
TCPProxy::summary           status:On
TCPProxy::emails            status:On

Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/package/TCP-Proxy.py", line 222, in run
    self.main()
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/package/TCP-Proxy.py", line 265, in main
    q = Queue.Queue()
AttributeError: class Queue has no attribute 'Queue'
SSLstrip v0.9 + POC by Leonardo Nve
-------------------------------
Thread::[DHCPServer] successfully stopped
Thread::[StaticHostapd] successfully stopped.
Thread::[DHCPServer] successfully stopped
Thread::[DNSMasq Server] successfully stopped.
Thread::[TCP Proxy] successfully stopped.
Thread::[SSLStrip+DNS2Proxy] successfully stopped.
[*] Configuring Static AP Mode...
[*] Enable forwarding in iptables...
-------------------------------
AP::[PumpkinAP] Running...
AP::BSSID::[bc:f6:85:21:26:0b] CH 11

[*] Loading debugging mode
[*] Current Session::ID [ODU5NTI=]
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/utility/threads.py", line 272, in run
    reactor.run(installSignalHandlers=False)
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/base.py", line 1260, in run
    self.startRunning(installSignalHandlers=installSignalHandlers)
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/base.py", line 1240, in startRunning
    ReactorBase.startRunning(self)
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/base.py", line 748, in startRunning
    raise error.ReactorNotRestartable()
twisted.internet.error.ReactorNotRestartable
[*] Sharing Internet Connections with NAT...
[New Thread 6148 (StaticHostapd)]
[New Thread 6150 (DNSMasq Server)]

[*] TCPProxy running on port 80/8080:

TCPProxy::ftp               status:On
TCPProxy::hexdump           status:On
TCPProxy::kerberos          status:On
TCPProxy::httpCap           status:On
TCPProxy::imageCap          status:On
TCPProxy::summary           status:On
TCPProxy::emails            status:On

Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/package/TCP-Proxy.py", line 222, in run
    self.main()
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/package/TCP-Proxy.py", line 265, in main
    q = Queue.Queue()
AttributeError: class Queue has no attribute 'Queue'
SSLstrip v0.9 + POC by Leonardo Nve
```
If you want the log.

mh4x0f, 2019-03-08:
the problem is this : 
[New Thread 5613 (DNSMasq Server)]
the DNSmasq server is not to be use, I 'll lock this. use the pyDnsServer change this on Settings

mh4x0f, 2019-03-08:
@minhducsun2002  another problem is the module Queue, I 'll check this but the error not exerts influence on internet connection

minhducsun2002, 2019-03-08:
@mh4x0f I will report results by tomorrow (UTC+7) 'cause I have works right now; feel free to focus on other things.

mh4x0f, 2019-03-08:
@minhducsun2002 nice, I 'll create a chat discussion the WP on discordapp for community

minhducsun2002, 2019-03-09:
[image]

This is a big problem indeed.

minhducsun2002, 2019-03-09:
[image]

As you requested.

mh4x0f, 2019-03-09:
set in DHCP tab to pyDHCPServer

minhducsun2002, 2019-03-09:
> set in DHCP tab to pyDHCPServer

I did.

DNSMasq DHCP even breaks WiFi-Pumpkin itself.

mh4x0f, 2019-03-09:
send me the output the terminal with this configuration

minhducsun2002, 2019-03-09:
For the sake of testing, I disabled the proxy.
```
cipher@magister-mentium [0] ~/Programs/Pen-testing/WiFi-Pumpkin $ sudo wifi-pumpkin
:0: UserWarning: You do not have a working installation of the service_identity module: 'cannot import name opentype'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
Loading GUI...
libpng warning: iCCP: known incorrect sRGB profile
WiFi-Pumpkin Running!
[*] Configuring Static AP Mode...
[*] Enable forwarding in iptables...
-------------------------------
AP::[cipPumpAP] Running...
AP::BSSID::[50:3e:aa:ae:a0:c8] CH 11

[*] Loading debugging mode
[*] Current Session::ID [MjQxMzU=]
[New Thread 12073 (StaticHostapd)]
[New Thread 12088 (No Proxy)]

```

mh4x0f, 2019-03-09:
did you try to reinstall the Tool ?, I don't find anything to reproduce this bug

minhducsun2002, 2019-03-10:
@mh4x0f How do you build Debian package? I'll make a build and overwrite your existing stable build to pick up your changes.

mh4x0f, 2019-03-10:
@minhducsun2002 I use the make_deb.sh for build .deb.

minhducsun2002, 2019-03-10:
Okay, got it, will build as soon as I get my PC back (currently on phone).

mh4x0f, 2019-03-14:
@minhducsun2002 did you got some result ?

minhducsun2002, 2019-03-14:
@mh4x0f Sorry for late reply - same problem, even with newly made Debian package :(

closed 2019-03-23 by mh4x0f