P0cL4bs/WiFi-Pumpkin-deprecated #475: DNSMasq DHCP error

H4rl3q1n, 2019-03-09:
When I select DNSMasq DHCP Server and start, I get an error message
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 816, in start_access_point
    self.dhcpcontrol.Start()
  File "/usr/share/WiFi-Pumpkin/core/controllers/dhcpcontroller.py", line 14, in Start
    self.Active.Start()
  File "/usr/share/WiFi-Pumpkin/core/servers/dhcp/dhcp.py", line 45, in Start
    self.Initialize()
  File "/usr/share/WiFi-Pumpkin/core/servers/dhcp/DNSMasq.py", line 21, in Initialize
    with open(leases, 'wb') as leaconf:
TypeError: an integer is required

I do not know how to fix that. I would use the Python DHCP which works perfectly, but I need the DHCP to be a specific range and I cannot change these settings for the Python DHCP. It is always the network 10.0.0.0

Help would be appreciated

mh4x0f, 2019-03-09:
nice report,  @H4rl3q1n  I 'll check this coming soon ;)

mh4x0f, 2019-03-10:
@H4rl3q1n I 'll disable the DNSmarq server and fix the bug from pyDHCP server.

closed 2019-03-11 by mh4x0f