P0cL4bs/WiFi-Pumpkin-deprecated #480: CredMonitor not capturing with Phishing Manager

Denzel-Washington, 2019-03-18:
**Description of the bug or feature request**

When using phishing manager with a custom template with login form like username / password
CredMonitor is not capturing any inputs. do take note before you close this issue and set to invalid, your previous versions was working fine with this method.

**To Reproduce**
Steps to reproduce the behavior:
1. start ap
2. start dns spoofer
3. start phishing manager
4. set directory to var/ww/html/

**Expected behavior**
credmonitor should be able to capture login inputs ..

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
* Version used tool: 0.8.7
* Virtual Machine (yes or no and which): no
* Operating System and version: parrot

** INCOMPLETE REPORT WILL BE CLOSED **

jeretc, 2019-04-19:
Use 0.8.4 ver, the NetCreds Logger works with custom user login portals.

jepunband, 2019-05-03:
> Use 0.8.4 ver, the NetCreds Logger works with custom user login portals.

Thanks. It works on this old version.