P0cL4bs/WiFi-Pumpkin-deprecated #483: Arch Linux package

WyRe, 2019-03-30:
**Description of the bug or feature request**

There is not bug to describe, this is only a question about Arch Linux chances to get a proper package for your software, I've found this one in [AUR, wifi-pumpkin-git](https://aur.archlinux.org/packages/wifi-pumpkin-git/) but like you can see it is totally wrong, it is not packaging anything at all. I was wondering if the maintainer/creator could be among you. 

Of course you are free to close this if you feel this is not the proper place to ask for this.

bionade24, 2019-03-30:
It's in the blackarch repo, maybe add this to README: https://github.com/P0cL4bs/WiFi-Pumpkin

WyRe, 2019-03-30:
Well, I'll keep open until devs add that info to the README, 

Thank you, @bionade24

mh4x0f, 2019-03-31:
WiFi-Pumpkin only make support to systems based debian.

closed 2019-04-02 by mh4x0f