P0cL4bs/WiFi-Pumpkin-deprecated #484: Switching to PyQt5

bionade24, 2019-03-31:
python2-sip-pyqt4 isn't supported properly anymore, packaging only works with legacy dependencies. Switching to PyQt5 is necessary.

mh4x0f, 2019-04-08:
for now, I `m focus in port the WP to python 3 version. change this now is a big problem because is not simply change the import pyqt4 to pyqt5. pyqt5 is very cool, but come many incompatibility with pyqt4. I think that this is possible when the WP is running on python 3.

blshkv, 2019-05-03:
somebody should have listened and do all migrations much earlier:
https://github.com/P0cL4bs/WiFi-Pumpkin/issues/174
``` not possible to implement, the project got too big for change"
:Facepalm

mh4x0f, 2019-05-24:
[image]