P0cL4bs/WiFi-Pumpkin-deprecated #488: service_identity module and other errors

Throwaway684736, 2019-04-08:
Raspbian Stretch, running the latest Pumpkin 0.8.8

uname -a
Linux localhost 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

I have this module installed:

pip install service_identity  
Requirement already satisfied: service_identity in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied: cryptography in /usr/lib/python2.7/dist-packages (from service_identity)
Requirement already satisfied: pyasn1-modules in /usr/local/lib/python2.7/dist-packages (from service_identity)
Requirement already satisfied: attrs>=16.0.0 in /usr/local/lib/python2.7/dist-packages (from service_identity)
Requirement already satisfied: pyasn1 in /usr/lib/python2.7/dist-packages (from service_identity)
Requirement already satisfied: ipaddress; python_version < "3.3" in /usr/lib/python2.7/dist-packages (from service_identity)
=================================================

But there's still an error:

# python2.7 wifi-pumpkin.py
:0: UserWarning: You do not have a working installation of the service_identity module: 'cannot import name opentype'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 36, in <module>
    from core.loaders.checker.networkmanager import CLI_NetworkManager,UI_NetworkManager
  File "/root/tools/WiFi-Pumpkin/core/loaders/checker/networkmanager.py", line 3, in <module>
    from core.main import Initialize,QtGui
  File "/root/tools/WiFi-Pumpkin/core/main.py", line 33, in <module>
    from core.controllers.proxycontroller import *
  File "/root/tools/WiFi-Pumpkin/core/controllers/proxycontroller.py", line 4, in <module>
    from core.servers.proxy.package import *
  File "/root/tools/WiFi-Pumpkin/core/servers/proxy/package/TCP-Proxy.py", line 7, in <module>
    import queue
ImportError: No module named queue

Throwaway684736, 2019-04-08:
I expect this ticket to be closed without any help from the author.
I ask other users to post their solutions if they can. Thanks.

jeretc, 2019-04-09:
> I expect this ticket to be closed without any help from the author.
> I ask other users to post their solutions if they can. Thanks.

Try #346

jepunband, 2019-04-13:
hi i'm having the same issue with the latest wifi-pumpkin, previous version was working fine on Parrot OS.
Regretting installing the latest version... : (

closed 2019-05-02 by mh4x0f