P0cL4bs/WiFi-Pumpkin-deprecated #491: no internet on access point

timhealltheway, 2019-04-18:
Hi,

No internet on access point. I tried to solve it from other issues above.

Here is my log when I am running Ap

```
```
root@kali:~# sudo iptables -t nat -L
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination         

Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination         
MASQUERADE  all  --  anywhere             anywhere   

root@kali:~# sudo iptables -t nat -L && ifconfig
Chain PREROUTING (policy ACCEPT)
target     prot opt source               destination         

Chain INPUT (policy ACCEPT)
target     prot opt source               destination         

Chain OUTPUT (policy ACCEPT)
target     prot opt source               destination         

Chain POSTROUTING (policy ACCEPT)
target     prot opt source               destination         
MASQUERADE  all  --  anywhere             anywhere            
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 10.5.49.26  netmask 255.255.248.0  broadcast 10.5.55.255
        inet6 fe80::27b8:6b4e:4a1d:e6c5  prefixlen 64  scopeid 0x20<link>
        inet6 2604:fc00:c:2525:a9ce:7610:d606:3ccd  prefixlen 64  scopeid 0x0<global>
        ether 64:00:6a:8a:48:8f  txqueuelen 1000  (Ethernet)
        RX packets 318906  bytes 293075606 (279.4 MiB)
        RX errors 0  dropped 853  overruns 0  frame 0
        TX packets 89720  bytes 8610987 (8.2 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 20  memory 0xf7c00000-f7c20000  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 600  bytes 70780 (69.1 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 600  bytes 70780 (69.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1400
        inet 172.16.0.1  netmask 255.255.248.0  broadcast 172.16.7.255
        ether bc:f6:85:21:26:0b  txqueuelen 1000  (Ethernet)
        RX packets 5085  bytes 575277 (561.7 KiB)
        RX errors 0  dropped 17  overruns 0  frame 0
        TX packets 1595  bytes 299972 (292.9 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```
**Here is my setting:**

[image]
[image]

iAMxGiGO, 2019-04-24:
same issue running wifi-pumpkin v0.8.8

jimknoepfle, 2019-04-27:
Hi, i have the same issue. Tried this on two different Raspi 3 B+ and on a laptop. Tried also different Network Connection and modified DHCP as mentioned before. 
two use a complete non protected WiFi enviroment i used 4G router. However i could resolve the issue.

any idea is apreciated

JimK

mh4x0f, 2019-05-26:
@timhealltheway do you using router ? because the IP address is not comum range of IP address, I think that the problem is come from your provider

timhealltheway, 2019-05-26:
@jimknoepfle @mh4x0f 
Hey guys, i finally found issues that school wifi is kind of tricky. they only allow one PC to have one Mac address.However, it works fine at home wifi.

mh4x0f, 2019-05-26:
ok, nice report ;)

closed 2019-05-28 by mh4x0f