P0cL4bs/WiFi-Pumpkin-deprecated #496: Issue with pumpkin proxy for enable/disable plugin options

golost2202, 2019-04-25:
When i set up my AP, every time i want to enable or disable some plugins options in the "pumpkin proxy" tab an error message popup in the terminal :

Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/package/PumpkinProxy.py", line 187, in setPluginOption
    self.reactor.ThreadPumpkinProxy.m.disablePlugin(name, status)
AttributeError: 'ThreadPumpkinProxy' object has no attribute 'ThreadPumpkinProxy'

For reproduce, open Wifi pumpkin, enable the pumpkin proxy and try to enable or disable some plugins

* Version used tool: 0.8.8
* Virtual Machine (yes or no and which): Yes with oracle VirtualBox
* Operating System and version: Kali Linux 2019.1

r3vy, 2019-06-13:
Can confirm. I cant figure this issue out. Im on Parrot Security OS, I even did a complete fresh install of my OS.

fawaz1984, 2019-11-22:
???