P0cL4bs/WiFi-Pumpkin-deprecated #499: self-update must be optional

blshkv, 2019-05-03:
If installed as a system tool under any Linux distro, the self-update function will be treated as  malicious behaviour. Additionally, the tool should not attempt to make external connections during the startup.

A linux package manager records location and checksum of all installed files and self-update will break it. 

Please make it optional via --update.
You can also try to guess and suggest updating the tool if it is older than xx days, like it was done in sqlmap https://github.com/sqlmapproject/sqlmap/issues/3553 recently.

mh4x0f, 2019-05-28:
I check the code in sqlmap upadate version, he using git for make update. When the user make a git pull is (function will be treated as malicious ) ?

blshkv, 2019-05-28:
can you point it out in the source code? I doubt this is the case.
You can make a call to check if the current version is the latest, but should not update anything.

mh4x0f, 2019-05-28:
look, [sqlmap core update](https://github.com/sqlmapproject/sqlmap/blob/master/lib/core/update.py#L109]). In case of WP, I force the download if the user is not installed the git tool, maybe this is the problem about you want to say. Remove this option and force the user use the git for update is a great option for me, but many users using the download in .zip on repo or install using .deb. btw, if use the --update or not, the tool will be run another process for execute the git pull, as you can see in core the update from sqlmap.

blshkv, 2019-05-28:
```
[20:33:55] [WARNING] not a git repository. It is recommended to clone the 'sqlmapproject/sqlmap' repository from GitHub (e.g. 'git clone --depth 1 https://github.com/sqlmapproject/sqlmap.git sqlmap')
do you want to try to fetch the latest 'zipball' from repository and extract it (experimental) ? [y/N] y
[20:33:57] [ERROR] unable to update content of directory '/usr/share/sqlmap' ('[Errno 13] Permission denied: u'/usr/share/sqlmap/sqlmap.py'')
[20:33:57] [INFO] for Linux platform it's recommended to install a standard 'git' package (e.g.: 'sudo apt-get install git')
```

mh4x0f, 2019-05-28:
exactly, as I have commented for you if you not have git installed, the sqlmap go to make download .zip file (WP too), this is  the problem. [sqlmap download file zip](https://github.com/sqlmapproject/sqlmap/blob/master/lib/core/update.py#L74)

blshkv, 2019-05-28:
Ok, can you make the same option?