P0cL4bs/WiFi-Pumpkin-deprecated #502: vpn

oakes8585, 2019-05-26:
**Fails to have connection on cilent wile vpn connected**

perfect connection with out vpn connected, would there be any way to just add the option to use tun0 for connection i read another issue ticket --https://github.com/P0cL4bs/WiFi-Pumpkin/issues/359-- where it was stated openvpn was blocking requests via wifi-pumpkin so just a switching option of the interface or the option  of choosing the local ip such as 10.34.0.14 etc should fix the issue a drop down menu on the bottom left of wifi pumpkin where it shows connection with those options would be best suited i think???

Thank you

* Version used tool: 0.8.8
* Virtual Machine no
* Operating System and version: kali linux 2019.2

mh4x0f, 2019-05-27:
@oakes8585 work fine with tun0 interface ? did you tested ?

oakes8585, 2019-05-28:
unless you have updated since i made this post it wouldent work for me i connected to vpn ran pumpkin and zero connection as soon as i disconnect then restart the application works like a charm

mh4x0f, 2019-05-28:
if not work because you open a new issue ? In #359  I ask that this is not a problem with WP. I recommend that you search on google.

closed 2019-05-28 by mh4x0f

oakes8585, 2019-05-28:
Ok ill try and figure out my issue my vpn blocks all connections that are not running through it or refuse to so i suppose i need to use iptables to adjust my issue.

Thanks for your time