P0cL4bs/WiFi-Pumpkin-deprecated #504: Wireless operation mode

zeraoo98, 2019-06-06:
I have these buttons grayed out, even tho all dependencies are installed and working separately, am i missing something here?

[image]

Version used tool: 0.8.8
Virtual Machine (yes or no and which): no
Operating System and version: Kali 2019.2

mh4x0f, 2019-06-08:
Hi, first of all, it hard for me add support for all OS to compile Mana and Karma Hostapd version, eve more so when it comes the OS Rolling release. we tools need only the binary compiled to run the wireless mode custom. btw, to enable this option only to copy the binary hostapd compiled to the folder  WiFi-Pumpkin/plugins/bin. 
[image]

obs:
I recommend that you search on google "how to compile the hostapd Karma or Mana" and copy the binary generated to folder as you can see on screenshot above. if you make this, the options **Mana AP Mode** and **Karma AP mode** will be enable when you restart the application again. 

Have fun!!

mh4x0f, 2019-06-08:
the wiki the hostapd mana is great look 
https://github.com/sensepost/hostapd-mana/wiki

zeraoo98, 2019-06-10:
Thank you for your reply, I will make sure to try your advice next time I'm on the machine.

closed 2019-07-31 by mh4x0f