P0cL4bs/WiFi-Pumpkin-deprecated #508: Cannot complete clean install (Kali on Pi3)

damienkeffyn, 2019-06-20:
I have a fresh install of Kali, curent version for RasPi

I have followed the instructions for down load but get no luck when trying to access through the AP I start, I can connect to it, but no internet access.

There is an issue with the install I cant answer, I have manually installed "pillow" but I cannot install through the script lxml. I gets to this point and literally freezes the Pi, only way out pull the plug and reboot.

I have manually installed the lxml with "pip install lxml" and "pip install pillow" both which completed successfully but when I re run the installer.sh this (below output) happens, it gets the "Running setup.py bdist_wheel for lxml ... |" and it locks the entire Raspberry Pi.

I have reflashed the SDcards and tried 3 Pi's same result.

**To Reproduce**
Collecting cssutils<1.1,>=1.0.1 (from mitmproxy==0.18.2)
Collecting typing==3.5.2.2; python_version == "2.7" (from mitmproxy==0.18.2)
Requirement already satisfied: certifi>=2015.11.20.1 in /usr/lib/python2.7/dist-packages (from mitmproxy==0.18.2) (2018.8.24)
Requirement already satisfied: ipaddress<1.1,>=1.0.15; python_version == "2.7" in /usr/lib/python2.7/dist-packages (from mitmproxy==0.18.2) (1.0.17)
Requirement already satisfied: idna>=2.0 in /usr/lib/python2.7/dist-packages (from cryptography<1.6,>=1.3->mitmproxy==0.18.2) (2.6)
Requirement already satisfied: cffi>=1.4.1 in /usr/local/lib/python2.7/dist-packages (from cryptography<1.6,>=1.3->mitmproxy==0.18.2) (1.12.3)
Requirement already satisfied: setuptools>=11.3 in /usr/lib/python2.7/dist-packages (from cryptography<1.6,>=1.3->mitmproxy==0.18.2) (40.8.0)
Requirement already satisfied: itsdangerous>=0.21 in /usr/lib/python2.7/dist-packages (from Flask<0.12,>=0.10.1->mitmproxy==0.18.2) (0.24)
Requirement already satisfied: Werkzeug>=0.7 in /usr/lib/python2.7/dist-packages (from Flask<0.12,>=0.10.1->mitmproxy==0.18.2) (0.14.1)
Requirement already satisfied: Jinja2>=2.4 in /usr/lib/python2.7/dist-packages (from Flask<0.12,>=0.10.1->mitmproxy==0.18.2) (2.10)
Collecting singledispatch (from tornado<4.5,>=4.3->mitmproxy==0.18.2)
  Using cached https://files.pythonhosted.org/packages/c5/10/369f50bcd4621b263927b0a1519987a04383d4a98fb10438042ad410cf88/singledispatch-3.4.0.3-py2.py3-none-any.whl
Collecting backports-abc>=0.4 (from tornado<4.5,>=4.3->mitmproxy==0.18.2)
  Using cached https://files.pythonhosted.org/packages/7d/56/6f3ac1b816d0cd8994e83d0c4e55bc64567532f7dc543378bd87f81cebc7/backports_abc-0.5-py2.py3-none-any.whl
Collecting argh>=0.24.1 (from watchdog<0.9,>=0.8.3->mitmproxy==0.18.2)
  Using cached https://files.pythonhosted.org/packages/06/1c/e667a7126f0b84aaa1c56844337bf0ac12445d1beb9c8a6199a7314944bf/argh-0.26.2-py2.py3-none-any.whl
Collecting PyYAML>=3.10 (from watchdog<0.9,>=0.8.3->mitmproxy==0.18.2)
Collecting pathtools>=0.1.1 (from watchdog<0.9,>=0.8.3->mitmproxy==0.18.2)
Collecting editorconfig>=0.12.0 (from jsbeautifier<1.7,>=1.6.3->mitmproxy==0.18.2)
  Using cached https://files.pythonhosted.org/packages/57/a6/2ac764d383137fc73dbc532c9c09f98937d4c2197bb9c4067d747d45a3e1/EditorConfig-0.12.2-py2-none-any.whl
Collecting hpack<3,>=2.2 (from h2<2.5,>=2.4.1->mitmproxy==0.18.2)
  Using cached https://files.pythonhosted.org/packages/c0/df/2c35ce4aa80c5741f894532beed35ce521491164e3aca235890148f38304/hpack-2.3.0-py2.py3-none-any.whl
Requirement already satisfied: pycparser in /usr/local/lib/python2.7/dist-packages (from cffi>=1.4.1->cryptography<1.6,>=1.3->mitmproxy==0.18.2) (2.19)
Building wheels for collected packages: lxml, Pillow
  Running setup.py bdist_wheel for lxml ... |

closed 2019-06-25 by mh4x0f

damienkeffyn, 2019-06-25:
what does incomplete mean?

mlichtenwalner, 2019-07-05:
Install kali-linux-full and you'll get past that error it seems. I was having the same problem.

dxmasterlv, 2019-07-05:
```
  Using cached https://files.pythonhosted.org/packages/c0/df/2c35ce4aa80c5741f89                                                                        4532beed35ce521491164e3aca235890148f38304/hpack-2.3.0-py2.py3-none-any.whl
Collecting pycparser (from cffi>=1.4.1->cryptography<1.6,>=1.3->mitmproxy==0.18.                                                                        2)
  Using cached https://files.pythonhosted.org/packages/68/9e/49196946aee219aead1                                                                        290e00d1e7fdeab8567783e83e1b9ab5585e6206a/pycparser-2.19.tar.gz
Building wheels for collected packages: lxml, brotlipy, backports.ssl-match-host                                                                        name, tornado, pyperclip, watchdog, configargparse, Pillow, cssutils, typing, cf                                                                        fi, pathtools, pycparser
  Running setup.py bdist_wheel for lxml ... |

```
Same issue here.