P0cL4bs/WiFi-Pumpkin-deprecated #520: WIFI-PUMKIN wont connect to the internet.

Rdgeno, 2019-09-17:
In a previous post I included a log that showed no errors and explained that The Pumpkin says and shows its connected to the internet but when you connect to the ap there is no internet. I and a lot of other people have asked for a solution to this problem here and a lot of other places. The answers that can be found as I previously stated don't work. I can repost the log of anyone wants to see it but if you have a working version running on the newest Kali Linux 2019.3 or a version working as I have described just fire it up and take a look at the log it's has to be the same if it's not showing errors. Git hub can you please answer this question or leave it up so someone can answer it, closing the question isnt an answer and ignoring the issue doesn't fix it. If anyone out there does have an answer I and many others thank you in advance. Thank you for your time and Git hub hopefully I can thank you for not closing this issue without an answer.

nickster076, 2019-09-30:
+1   Tried everything  but cant acces internet even tho connected. Clients: 0 in pumpkin  but when i enter a website i can see it showing up in pumpkin but its broke AF

CharlieBrown805, 2019-11-19:
Ifconfig 
Copy paste the ssid to wifi-pumpkin. Clients connected to AP but I can't see there activity

Rdgeno, 2019-11-19:
See if it shut off your wifi some how it changes it from =managed to unmanaged. I gave up on it I installed it four times every time same problem and it would kill the wifi. I just tried it again and I have been trying to get my wifi to work correctly ever since try mitmap works flawlessly. For captive portal unfortunately airgeddon works the best it has horrible portals but it works properly. Right now I'm trying to learn how to make my own portals. At some point you have to progress from script to doing it yourself if not your never going to get anywhere. A lot if this stuff is outdated by the time it's a script. We arent getting the good stuff the ninjas that really do this stuff dont let that be known as soon as words out patches. Try Mitmap if you just want to mitm if you want a captive portal airgeddon. Airgeddon has evil twin and stuff but it doesnt like the newest version of some program I cant remember so use mitmap for that stuff it automates everything. Just answer the questions hit enter and your up and running.

CharlieBrown805, 2019-11-21:
It didn't shut off my internet.  Only thing is I can't see the clients.  Can be right about the outdated thing but I also just create my AP using oblique/AP it work with in minutes.  Thanks man I'll try those things. I think also with more knowledge of the specific tools it would be easier

Rdgeno, 2019-11-21:
Your welcome I kind of misspoke though the USB adaptor was affected. If found in the NetworkManager folder the .conf file was changed, but yeah it killed the adaptor. The first few times I tried it it wouldn't connect like your saying then after a couple months I tried it again the other day and everything worked perfectly till I shut it off I only had the onboard wifi again no adaptor. I'm glad you found something that works for you good luck.