P0cL4bs/WiFi-Pumpkin-deprecated #521: Pumpkin won't start

patuuh, 2019-09-27:
When trying to run wifi-pumpkin in terminal, I get this error: 
```
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 36, in <module>
    from core.loaders.checker.networkmanager import CLI_NetworkManager,UI_NetworkManager
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/networkmanager.py", line 3, in <module>
    from core.main import Initialize,QtGui
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 20, in <module>
    from core.helpers.update import frm_githubUpdate
  File "/usr/share/WiFi-Pumpkin/core/helpers/update.py", line 5, in <module>
    from core.loaders.master.github import GithubUpdate,UrllibDownload
  File "/usr/share/WiFi-Pumpkin/core/loaders/master/github.py", line 1, in <module>
    from config import Config
ImportError: cannot import name Config
```
When I tried to fix this, it gives me another different error after another.
I used the normal installation method (./installer.sh --install) and it finished without an error.

I'm running on Kali linux 2019.4

anonymous010010, 2019-10-02:
Having the same issue, did you fix this?

hackvice, 2019-12-11:
When i try to start mine it asks me for password, and when i provide my password says its not the same 0.0