P0cL4bs/WiFi-Pumpkin-deprecated #522: Sudo: pip2: Command not found

janak6625, 2019-10-05:
I am Trying to install Wifi pumpkin 2 , on step 2 i get this error "Sudo: pip2: Command not found"
Note: I am noobie and just install Kali linux First time (I ve never install Linux in my life)

**To Reproduce**
Steps to reproduce the behavior:

I am trying to follow installation procedure below on my Kali linux: 
Installation procedure

 1) git clone https://github.com/P0cL4bs/WiFi-Pumpkin.git  (I have success in this step)
 cd WiFi-Pumpkin (Success )
 sudo pip2 install -r requirements.txt (I get error " Sudo: pip2: Command not found")
 chmod +x installer.sh (not yet there)
 sudo ./installer.sh --install (not yet there)

Find Screen shot : [https://snipboard.io/al1LCI.jpg](url)

**Desktop (please complete the following information):**
* Version used tool: Wifi pumpkin : 0.8.8
* Virtual Machine (yes or no and which): not virtual machine
* Operating System and version: Linux kali 5.2.0-kali2-amd64 #1 SMP Debian 5.2.9-2kali1 (2019-08-22) x86_64 GNU/Linux

janak6625, 2019-10-07:
anyone ? can answer this question please ?

alloygoh, 2019-10-09:
@janak6625 It means that you do not have the `python2-pip` package installed. Install it by running `sudo apt install python2-pip`.

janak6625, 2020-01-04:
it says E: unable to locate package python2-pip

janak6625, 2020-01-11:
root@kali:~/WiFi-Pumpkin#  sudo pip2 install -r requirements.txt
sudo: pip2: command not found
root@kali:~/WiFi-Pumpkin#  pip2 install -r requirements.txt
bash: pip2: command not found
root@kali:~/WiFi-Pumpkin# cd
root@kali:~# sudo apt install python2-pip
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package python2-pip
root@kali:~#

janak6625, 2020-01-11:
why in this world Linux is tooooooo complicated

janak6625, 2020-01-11:
Update- 
https://askubuntu.com/questions/672808/sudo-apt-get-install-python-pip-is-failing
above post help me

closed 2020-01-11 by janak6625