P0cL4bs/WiFi-Pumpkin-deprecated #524: Problem installing mitmproxy==0.18.2

elaxrocha, 2019-10-09:
Problem installing mitmproxy==0.18.2

A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Open terminal and do `sudo pip install mitmproxy==0.18.2`

**Expected behavior**
Package would be installed and wifi-pumpkin "Pumpkin Proxy" would work

**Screenshots**
[image]

**Desktop (please complete the following information):**
* Version used tool: mitmproxy==0.18.2 
* Virtual Machine (yes or no and which):  no
* Operating System and version: Debian 10 buster

** INCOMPLETE REPORT WILL BE CLOSED **

elaxrocha, 2019-10-10:
*Update*
After uninstalling the last version of wifi-pumpkin and installing the version 8.5 i was able to open it and update via the application. Now I can use it without any problems.

cojovi, 2020-02-04:
ive done that, and still am getting the error "Package Requirement" "the package requirement mitmproxy==0.18.2 not satisfied"

any other options?